ChrGri/DIY-Sim-Racing-FFB-Pedal #28: can this design be compatible with this Nema 23 motor?

B3N1321, 2024-04-08:
Just found this amazing item on AliExpress. Check it out! HUF12,788.29  50％ Off | Nema23 Closed Loop Integrated Motor 1.2Nm 2Nm 2.5Nm 3Nm Diameter 8mm Hybrid Integrated Servo Motor with Drive For CNC
https://a.aliexpress.com/_EGOXpFh

B3N1321, 2024-04-08:
> Hi, whats the reason for this issue?

I found another type of engine that doesn't have the controller on the
side, so it's much more aesthetic, but I don't know if it can work with it.

Here is the link:

Just found this amazing item on AliExpress. Check it out! HUF12,788.29 50％
Off | Nema23 Closed Loop Integrated Motor 1.2Nm 2Nm 2.5Nm 3Nm Diameter 8mm
Hybrid Integrated Servo Motor with Drive For CNC
https://a.aliexpress.com/_EyO5spp

and I want to join the development group

Thank you
Benjamin

ChrGri ***@***.***> (időpont: 2024. ápr. 8., H, 20:36) ezt
írta:

ChrGri, 2024-04-08:
It wouldn't work. The proprietary data protocol of the iSV57 servo was decoded/reverse engineered and is a substantial part of the code. 
Furthermore, IIRC, this motor has been tested by someone at the beginning of the project and found not be sufficient. More info can be found on the discord channel.
BR

closed 2024-04-08 by ChrGri