ChrGri/DIY-Sim-Racing-FFB-Pedal #41: iHSV57 compatibility (question)

Javiju555, 2024-10-20:
Hello, I'm wondering if it could work with iHSV servos like this:
https://a.aliexpress.com/_EunFoh7
I know you already told that it only works with iSV proprietary protocols, but at least for me that iHSV servos looks similar to the iSV.
Thanks

ChrGri, 2024-10-24:
I cannot support the implementation to make this motor work with the code. It would be up to you to identify the servos registers, tune the motor parameters and check the performance.

Would you see yourself doing that? 

BR
Christopher

Javiju555, 2024-10-24:
Well, maybe I could follow the steps that were made for analyze the iSV for that one but since I haven't bought any for now I will probably buy the stepper online iSV, I don't have that much time to identify the parameters and tune it in short-term...

ChrGri, 2024-10-24:
Ok. I'll close the ticket for now. Feel free to reopen the ticket, if you change your mind.

closed 2024-10-24 by ChrGri