ChrGri/DIY-Sim-Racing-FFB-Pedal #49: PCB V4 stable USB connection

xxLEITTOxx, 2024-12-20:
Hey man :) I think title says it all.
Original quote "I'm totally ok with two usb cables 🙂 that would be awesome if it's going to be stable connection with usb-hid. Had some experience with vjoy and want to avoid it at all cost. If it's going to help in development - you can add some type of cod to gather data from working pedal. I'd be happy to share automatic feed. I will agree to smth like this - LEITTO pedal been working for 3hours, servo temp was 98C and servo control board gone in protection. Imho thats will be valuable data 🙂"
and "You'd do me a favour by opening a git issue for that matter. Otherwise, I might forget that stuff."
Thank you man :)

ChrGri, 2024-12-23:
Code changes have been implemented with https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/commit/3af58c503f8b0f0b740b72c302af1aab70f109df. Will be available in the next release.

Tests are good so far. No connection loss.

BR

ChrGri, 2025-02-19:
CDC has been deactivated. Serial and HID output via different ports has been activated. V5 PCB was released. Therefore closing the issue.

BR

closed 2025-02-19 by ChrGri