ChrGri/DIY-Sim-Racing-FFB-Pedal #50: engine overheating randomly without engine error code

Vins34130, 2024-12-23:
Hello,

As seen on the discord, my engine heats up randomly, last night for 30 minutes it went into safety mode, whereas last Thursday I did a 1h15 race without any problem and without heating up.

Thx

Vins34130, 2024-12-23:
I'm going to play for 1 hour this afternoon, I'll tell you.

ChrGri, 2024-12-23:
Could you please record and provide the pedal state log. Possibly when the overheating occurs.

BR

Camel97, 2024-12-25:
I'm here with the same Problem, but it is rarely. I log pedal states and hope that the issue could be recorded.

ChrGri, 2024-12-25:
@Camel97 and @Vins34130, the iSVs internal safety alarms have been reactivated since [this pre release](https://github.com/ChrGri/DIY-Sim-Racing-FFB-Pedal/releases/tag/Develop_Package_v20241214.032245). Please update to the latest prerelease before continuing.

BR
Chris

Camel97, 2024-12-26:
70 degrees after 1 hour 
[Uploading output_1.txt…]()

Vins34130, 2024-12-26:
I'm on vacation, I can only try this evening or tomorrow.

ChrGri, 2024-12-26:
> 120 degrees on motor after 1 hour Uploading output_1.txt…
> 
> Fan of the Power Supply is running High.
> 
> Hardware: Meanwell 350w 36VDC Waveshare EPS32-S3 Loadcell Dyly 107 Motor ISV57T Nema23 130W GX1610 (100mm) I'm only using 40mm.
> 
> Prelease ist not installed!

The log output seems to be missing.

BR
Chris

Vins34130, 2024-12-26:
I updated with the dev but the pedals shake as soon as I press on them, or even micro movements alone.

ChrGri, 2024-12-26:
> I updated with the dev but the pedals shake as soon as I press on them, or even micro movements alone.

There were two bugs in the MPC code, which makes retuning the controller necessary. Please try a MPC value of 1.5kg/mm or one of my online configs.

BR
Chris

Vins34130, 2024-12-26:
online config no work
[image]

Camel97, 2024-12-26:
[Uploading output_1.txt…]()

ChrGri, 2024-12-26:
> online config no work [...]

The SimHub plugin might not be updated in the current release. Alternatively the config can be found [here](https://discord.com/channels/1113129142142120159/1173200481636589579/1317818512244346900).

BR
Chris

ChrGri, 2024-12-26:
> Uploading output_1.txt…

Still only a link to this issue.

BR
Chris

Vins34130, 2024-12-26:
Thanks the setting is perfect, now I will check if the engine heats up

ChrGri, 2025-02-19:
> Thanks the setting is perfect, now I will check if the engine heats up

Any updates?
BR

Vins34130, 2025-02-19:
Hi, since the update given above I no longer have any personal problems.

closed 2025-02-19 by ChrGri